st-one-io/node-red-contrib-s7 #98: S7 in error blocks entire workflow

mhechthz, 2021-08-31:
Hi,

with the 2.2.1 version I get a read error from an S7 (probably something wrong there). 

The node shows a yellow light and the info "fehlerhaft". On the debug console I get at every flow (even if on "display only active workflow") the message "Fehler (ungültige Werte)" (Error - invalid values). I'm not sure if the German writing is from my system or from the SPS configuration or from else where.

The catch node in the workflow doesn't catch anything and another watchdog node, that I use for a long time to check if nodes produce output or not seems to hang, so the entire workflow seems to be blocked by the S7 in node.

Any suggestions?

thermseekr, 2022-05-30:
Hello all. The config node should propagate the error to the visible node so the catch node could... catch it. Here is another user having the same problem:

https://discourse.nodered.org/t/how-to-catch-config-node-error/52573

I'm currently not skilled enough to do it, maybe someone with the right skills could fix this and propose a PR?

ajeschmec, 2022-09-22:
I experienced the same behavior, and also other error messages as "Error: connect ECONNREFUSED .. " are not propagated to catch nodes. The error remains after updating to version 3.1.0. 
Could someone fix this issue, please? 
It will be very helpful.

OriolFM, 2023-09-29:
Any updates on this? I'm having exactly the same issue. The visible node (s7 in) shows the offline status, which means the endpoint node is propagating the error to the visible node, but it does not generate any error message (on the s7 in node side of things).

The errors, in the meantime, are still uncatchable.