st-one-io/node-red-contrib-s7 #102: Multiple connections to plc when deploy instance

avlpjwt, 2022-01-18:
### Describe the bug
Every time we deploy the instance, it creates a new connection to plc and when it gets more than 32conn plc stops with conn limit

### To Reproduce
contact us at: marcelino.eilert@volkswagen.com.br

### Expected behavior
reset the connection to plc when deploy de instance

Dabbadoeber, 2022-04-11:
Hi i have the Same issue with an IM151-8
No connection Anymore, even TIA is unable to connect, i think probably al Connection resource are full.
is it possible to flush these ?

fernandoamorim1703, 2022-06-13:
the issue of the limit of simultaneous connections is on the part of the PLC that has a defined maximum, usually they are between 4 to 64 connections depending on the type of PLC

matthew-larner, 2022-09-30:
I'm also seeing a similar issue when it fails to connect to the PLC - every time it retires, it opens another connection without closing the previous.