st-one-io/node-red-contrib-s7 #104: Pn-tools

Ivan0x17, 2022-02-22:
What is pn-tools?

Directory

/src/tools.js

Line 12 = const pnToolsPath

Sulomus, 2022-04-21:
+1 I would love to know too!

I am in need to use PROFINET with diffrent kind of PLC then Siemens, and I would like to use the "core" of this project as a starting building block for it.

Does anyone have any idea what is "pn-tools"? @Ivan0x17 have you figured it out?

Ivan0x17, 2022-04-21:
> +1 I would love to know too!
> 
> 
> 
> I am in need to use PROFINET with diffrent kind of PLC then Siemens, and I would like to use the "core" of this project as a starting building block for it.
> 
> 
> 
> Does anyone have any idea what is "pn-tools"? @Ivan0x17 have you figured it out?

Unfortunately I don't know what it is pn-tools

fernandoamorim1703, 2022-06-13:
pn-tools is our internal tool to check how many and which PLCs are connected to the network

fernandoamorim1703, 2022-06-23:
We will be closing this issue, if there is a need to revisit the subject, please open a new one. Thank you very much.

closed 2022-06-23 by fernandoamorim1703