st-one-io/node-red-contrib-s7 #108: Cannot write Strings to PLC (S7-315 PN/DP)

ciphermaster, 2022-05-30:
### Describe the bug
The Node-Out Node in Version 3.1.0 does not support strings to be written to the PLC.
I am using a S7315 to write data to the HMI which was no issue in Version 2x

I have defined a String variable the following way:

STATUS    DB15,S564.254

When I try to use the inject node to provide a payload with the string "Running", the following error will be received:

[error] [s7 out:STATUS] Error: Cannot write item with size greater than max payload of [228]

Seems the node does not recognize the string lenght correctly when sending out.

fernandoamorim1703, 2022-06-23:
The problem lies in the size of memory space you are allocating to save the string "Running". When you declared DB15,S564.254 this value .254 is the memory space that will be used, 254 bytes.

Then it is necessary to decrease the size of bytes in the variable declaration to a smaller number.

PLCs have a maximum limit of bytes per transaction, this varies from one to another. So the space you are allocating is more than the CLP supports in the transaction.

In version 2.x, there was a treatment for when the transaction exceeded the CLP limit, breaking the transaction in parts so that it was successful. In version 3.x the implementation of the node was changed and this part of partitioning the transaction to not exceed the limit was not performed.

fernandoamorim1703, 2023-07-27:
We will be closing this issue, if there is a need to revisit the subject, please open a new one. Thank you very much.

closed 2023-07-27 by fernandoamorim1703