st-one-io/node-red-contrib-s7 #112: Change address from NodeRed property

gamesmusic454, 2022-09-29:
It would be pretty nice if we could change the IP address of the nodes sending the IP in msg.address for example, I've tried to modify the code but I did't have luck getting my own approach into NodeRed as I've never made a Node package, the code that I've modified it's in the S7.js and my approach is the following: `connOpts = {
                        host: msg.address,
                        port: Number(config.port),
                        rack: Number(config.rack),
                        slot: Number(config.slot),
                        s7ConnOpts: s7ConnOpts
                    }`
I hope that changing the `host:` value to `msg.address` gives you the possibility to make it. If there is anything required to make it work I could help with that.

fernandoamorim1703, 2023-07-27:
Hello @gamesmusic454 ,

Unfortunately, this change is not feasible, as the "config.host" parameter is used to retrieve information set in the endpoint (in the part that registers all the PLC information).

fernandoamorim1703, 2023-07-28:
We will be closing this issue, if there is a need to revisit the subject, please open a new one. Thank you very much.

closed 2023-07-28 by fernandoamorim1703