st-one-io/node-red-contrib-s7 #116: checkbox for suppressing "Error: connect EHOSTUNREACH" in debug sidebar of NodeRed

digidax, 2022-11-23:
**Is your feature request related to a problem? Please describe.**
When a PLC of a machine is not connected 24/7, it produces when requesting the IP of the PLC the "Error: connect EHOSTUNREACH". This is flooding the debug sidebar and also with the "catch node" the output can not be suppressed.
So when working in Node Red on weekend, where over 90% of the PLCs are offline, it's terrible to read or monitor a message from a projected debug node.

**Describe the solution you'd like**
A checkbox to disable / suppress this error messages.

**Describe alternatives you've considered**
Was discussed here [https://discourse.nodered.org/t/filter-errors-in-sidebar/71044](url) with no solution.

**Additional context**
[image]

1607David, 2023-05-19:
As far as i know its possible to catch error messages with an CATCH node, but only if the error function is feeded with two arguments.
node.error("Error : connect....",msg);
Maybe this could be changed ?

fernandoamorim1703, 2023-07-27:
Hello @digidax , 

 understand your concern, but the node was developed with machines that are operating 24/7 in mind, where any errors presented are of utmost importance, especially reconnection errors.

Adding a checkbox to suppress errors has a great potential for failure, as it would stop showing any errors that the PLC presents.

One solution would be for you to disable the endpoint when the PLC is not in operation.

ma-ze, 2023-08-03:
It is possible to select "filter this node" from one of the debug messages.
There is another issue though: I changed the name and IP of my S7 node, and the error messages of the old name and IP dont stop. I then deleted all nodes, even uninstalled contrib-s7 from my nodered, and the error messages are still coming. Seems like some process is still there trying to make a connection.

fernandoamorim1703, 2023-08-03:
Hello @ma-ze 
Did you check the configuration nodes? Perhaps they are still instantiated.

ma-ze, 2023-08-03:
Yes, after changing the configuration node (and then deleting it, and uninstalling the plugin) it still produces error messages.

digidax, 2023-08-04:
"Filter a node" to get the debug msg focus from a not connected S7 node doesn't solve the problem.
EHOSTUNREACH error msg are everytime present an flooding the debug output. That is my problem. We have a lot of machines which are only temp. in use. So I don't really understand, why would it not possible to make a checkbox for disabling this EHOSTUNREACH error msg.

its-Zeke, 2023-09-14:
+1 
I've got the same issue. Due to energy savings we turn off varies control units, which do not produce parts leading to error messages spamming my debug view. Either a checkbox to suppress something or at least a field to change the interval of the node trying to reconnect to the plc would be helpful.

AsepSubarkah, 2023-12-27:
Error: PLC error [0x8104]: This service is not implemented on the module or a frame error was reported

what is problem?

Peturtjuhh, 2024-01-26:
> Hello @digidax ,
>
> understand your concern, but the node was developed with machines that are operating 24/7 in mind, where any errors presented are of utmost importance, especially reconnection [...]

If that is not an option,
Could you point out how to turn off this error all together?
Wich module do i need to edit, and what line of it?

My thanks

digidax, 2024-05-27:
Since updating to nod-red v3.1.9, too many "EHOSTUNREACH" connect errors are production a restart of the complete nod-red instance:

`May 27 10:12:57 173 Node-RED[261193]: 27 May 10:12:57 - [error] [s7 endpoint:Presse 10] Error: connect EHOSTUNREACH 192.168.130.104:102
May 27 10:12:57 173 Node-RED[261193]: 27 May 10:12:57 - [error] [s7 endpoint:Presse 6] Error: connect EHOSTUNREACH 192.168.130.96:102
May 27 10:12:57 173 Node-RED[261193]: 27 May 10:12:57 - [error] [s7 endpoint:Presse 7] Error: connect EHOSTUNREACH 192.168.130.100:102
May 27 10:12:57 173 Node-RED[261193]: 27 May 10:12:57 - [error] [s7 endpoint:Presse 5] Error: connect EHOSTUNREACH 192.168.130.88:102
May 27 10:12:57 173 Node-RED[261193]: 27 May 10:12:57 - [error] [s7 endpoint:Presse 8] Error: connect EHOSTUNREACH 192.168.130.71:102
May 27 10:12:59 173 Node-RED[261193]: 27 May 10:12:59 - [red] Uncaught Exception:
May 27 10:12:59 173 Node-RED[261193]: 27 May 10:12:59 - [error] Error: read ECONNRESET
May 27 10:12:59 173 Node-RED[261193]:    at TCP.onStreamRead (node:internal/stream_base_commons:217:20)
May 27 10:12:59 173 systemd[1]: nodered.service: Main process exited, code=exited, status=1/FAILURE
May 27 10:12:59 173 systemd[1]: nodered.service: Failed with result 'exit-code'.
May 27 10:13:19 173 systemd[1]: nodered.service: Service RestartSec=20s expired, scheduling restart.
May 27 10:13:19 173 systemd[1]: nodered.service: Scheduled restart job, restart counter is at 5.
May 27 10:13:19 173 systemd[1]: Stopped Node-RED graphical event wiring tool.
May 27 10:13:19 173 systemd[1]: Started Node-RED graphical event wiring tool.
May 27 10:13:20 173 Node-RED[261588]: 27 May 10:13:20 - [info]
May 27 10:13:20 173 Node-RED[261588]: Welcome to Node-RED`

Referring to post [https://github.com/node-red/node-red/issues/4703](url) 
"The only practical fix is for the S7 node to handle it's errors properly. That isn't something the core of Node-RED can do for it.", we need please a solution for it, to switch off / on the error handling of EHOSTUNREACH please.

patrickschultes, 2025-06-26:
I just wanna add a new idea to the topic: 

Instead of "suppressing" the error it would be beautiful if it will be catchable by the "catch"-node. 
So in default the error will be thrown directly but the one who wanna do something with the information (or activly ignore them) would be able. 

As example when you need to restart the MPI to ETH gateway