st-one-io/node-red-contrib-s7 #120: Invalid properties S7 endpoint with nodered V3.0.2 Node.js 18.14.0 -Windows 10

Paolind, 2023-02-17:
Hi, the system gives me an error when I try to configure manually S7 endpoint to comand a light and a motorized roller shutter with nodered V3.0.2 Node.js 18.14.0 running with Windows 10. The strange thing is that when I import the same (two) flows from another Nodered instance (v.0.19.5) this problem doesn't appear for the light while persist for the motorized roller shutter. I double checked the differences between the two vesione of S7 endpoint but they seems to me filled with the same data and this  is corroborated from the fact that ONLY Once, when i use the imported flow for the motorized roller shutter in the windows machine, the shutter works, but then, when I try to move again the shutter, the system doesn't react. It is like the PLC accept only once the input and then fail. I am able to restart the system with the same flow using S7 with nodered instance v.0.19.5. Has anyone an Idea of the reason why the system react in this way? Thanks a million in advance, best regards. Paolo

tiscali19, 2023-06-11:
Hi, I have the same problem! Have a solution?

Paolind, 2023-06-11:
I’m out of home, furthermore after i fixed my problems I didn’t enter the system again, i don’t know if it helps you… try with the blue one.
regards
Paolo

Inviato da iPhone

tiscali19, 2023-06-11:
Paolo, are you italian?

Paolind, 2023-06-11:
si

closed 2023-06-11 by Paolind

tiscali19, 2023-06-12:
> si

Scusami, mi puoi dire più o meno cosa hai fatto? Sono bloccato...

Paolind, 2023-06-12:
Ciao, hai visto lo screenshot che ti ho mandato? ho installato il pacchetto cerchiato in verde dopo aver disinstallato il rosso

Inviato da iPhone

Paolind, 2023-06-12:
Blue non verde😅
Inviato da iPhone

tiscali19, 2023-06-12:
Non vedo nessuna immagine...

Paolind, 2023-06-12:
io ho installato il pacchetto il blue (dopo aver disinstallato quello in rosso)
giusto per curiosità fammi sapere se hai risolto

Inviato da iPhone

tiscali19, 2023-06-12:
Anche adesso non c'è nessuna immagine allegata. Mi puoi scrivere i nomi dei pacchetti?

Paolind, 2023-06-12:
mandami n cell ti mando wtsp

Inviato da iPhone

tiscali19, 2023-06-12:
+39 320 8576621

Paolind, 2023-06-12:
non ti trovo su wtspp

io avevo installato

plcindustry   (versione 2.1.1) 

e disinstallato 

node-red-contib-s7

Inviato da iPhone

tiscali19, 2023-06-12:
Ok, grazie, lo provo. Per Whatsapp hai scritto il numero giusto?