st-one-io/node-red-contrib-s7 #128: Error writing arrays

xrodriguez-dotcom, 2024-02-29:
### bug description

When trying to write an array, I get this error:  "Error: Number of tags must match the number of values"

### To Reproduce

1. Create an array variable.  e.g. DB1001,X0.0.10   (10 bit array variable)
2. Set up a S7 out node with this variable
3. pass the payload: [false,false,false,false,false,false,false,false,false,false] to the s7 out node
4. after the version 2.2.1 you will get the error "Error: Number of tags must match the number of values"

It was working fine in version 2.2.1

### Environment
 - **Version** of this node: [3.1.0]
 - **Node-RED version**: [3.1.5]
 - **PLC Part Number**: [6ES7 517-3FP00-0AB0 ]
 - **OS**: [Windows Server 2022]
 - **Browser**: [Edge]

gtf016, 2024-08-30:
使用[[false，false，false，false，false，false，false，false，false]]的形式在进行尝试一下看看

xrodriguez-dotcom, 2024-09-16:
> 使用[[false，false，false，false，false，false，false，false，false]]的形式在进行尝试一下看看

Thank you, but doing this I get  Error: Write error [0x7]: Data type inconsistent

Mustafa19077, 2025-05-10:
Hello rodrigues, could you solve it, if it is, how?